ibell/pdsim #6: Cannot load config file that has pressure under ptriple

Oberklasse, 2015-08-19:
Throws a ValueError: Pressure in PQ_flash [x Pa] must be in range [ y Pa, z Pa]

Traceback (most recent call last):
  File "PDSimGUI.py", line 1610, in OnConfigOpen
    self.rebuild(configdict)
  File "PDSimGUI.py", line 1014, in rebuild
    frame = MainFrame(configfile, position=position, size=size)
  File "PDSimGUI.py", line 854, in __init__
    self.build()
  File "PDSimGUI.py", line 1094, in build
    self.MTB = MainToolBook(self, self.config)
  File "PDSimGUI.py", line 795, in __init__
    self.InputsTB = family.InputsToolBook(self, configdict)
  File "C:\pdsim\GUI\families\scroll.py", line 186, in __init__
    pdsim_panels.StateInputsPanel(self, config['StatePanel'], name='StatePanel'),
  File "C:\pdsim\GUI\panels\pdsim_panels.py", line 2278, in __init__
    state = CP.State(inletState.Fluid, dict(P=pressure,Q=1))
  File "CoolProp\CoolProp.pyx", line 510, in CoolProp.CoolProp.State.__init__ (CoolProp\CoolProp.cpp:27176)
  File "CoolProp\CoolProp.pyx", line 604, in CoolProp.CoolProp.State.update (CoolProp\CoolProp.cpp:28521)
  File "CoolProp\AbstractState.pyx", line 39, in CoolProp.CoolProp.AbstractState.update (CoolProp\CoolProp.cpp:9199)

ibell, 2015-08-20:
Hmm yes I should definitely be able to fix this.  It would perhaps be good
to discuss in general a better way of dealing with the discharge state

ibell, 2015-12-11:
Closed by 7305a087e0ff591d5fa4f93a854a064b5f8e4380

closed 2015-12-12 by ibell