ibell/pdsim #21: Enable multiple lumped temperatures in the overall energy balance

davideziviani, 2016-06-27:
Allow to include detailed equivalent thermal resistances between different lumped elements especially for hermetic compressors. Example Chen PhD on scroll

ibell, 2016-06-28:
It's a good idea and should be mostly (completely?) implemented already.
What happens when you try to use multiple lumps?  Do you get an error?

davideziviani, 2016-06-28:
I looked at the code and it should be straightforward. Since we do not have an example of that capability, I plan to add it as I am working on a recip compressor and the detailed motor model is also available.

davideziviani, 2016-11-03:
Today I was trying to work on the multiple lumps as I wanna include this part into the second paper about PDSim. I found out there is an issue with assigning the number of lumps as well as the number of residuals (or equations) required. I am going to leave a comment at the proper line.

davideziviani, 2016-11-03:
The number of lumps need to match the number of residuals. In the current examples (recip,scroll, etc), lump_energy_balance_callback(self)  returns a single residual. So I tried to add two lumps (self.Tlumps[0], self.Tlumps[1]) and two residuals, but list index is out of range for self.Tlumps even if I pass a guess array x0 to self.solve( )

I think here [https://github.com/ibell/pdsim/blob/master/PDSim/core/core.py#L1118](url)  we need to generalize the structure to append multiple  [self.Tlumps, self.lumps_resid]. 

I will be working on this issue. let me know if you see a better way of handling this issue.

davideziviani, 2017-01-12:
It works fine at the moment also for speed. I am going to check if I need to implement a different algorithm besides Multi-Dim Newton Raphson.

davideziviani, 2020-03-27:
Solved in #39

closed 2020-03-27 by davideziviani