ibell/pdsim #24: Generalize friction corrected Isentr nozzle flow model

davideziviani, 2016-11-29:
Allow to specify the calibrated coefficients if needed to apply the flow model to different compressors/expanders

ibell, 2016-11-29:
Can you explain what you have in mind?

davideziviani, 2016-11-29:
Yes sure. I was working on the Z-compressor model and I am running the detailed flow model for two flow paths to determine the friction flow correction factor. As I am using the same fitting function you adopted in the hybrid leakage model, it is useful to enable the user to specify the a_0,...,a_10, Re_star coefficients.
Similar story applies for example to rolling piston compressor etc.

Does this make sense?

ibell, 2016-11-29:
I guess so, but that seems like a lot of adjustable parameters !  Would you
not be better off adjusting the flow path length or a general flow
coefficient?

davideziviani, 2016-11-29:
yup that is also true but in the case a user wanna use the friction corrected isentropic nozzle for a particular path of interest, currently it is not readily possible (let's say parametric studies on different path length and gap size). One has to change the coefficients first, and the recompile PDSim.

ibell, 2016-11-29:
That is true.  I wonder about the speed if you are always passing the
coefficients.  Probably better to come up with a callable class so the
coefficients can be cached.

davideziviani, 2016-11-29:
yes indeed. I am gonna take a look to create a callable class to cache the coefficients