ibell/pdsim #31: ForceFcn for pockets C1&2 are not well defined

sebdenis, 2019-04-12:
Hi,

there is an issue regarding the definition of `ForceFcn `methods for pockets C1 and C2 done in `pre_run` method in scroll.core. The code concerned is 
```
        for i in range(10):
            try:
                Fcn = lambda theta, geo: symm_scroll_geo.C1_forces(theta, i+1, geo)
                self.CVs['c1.' + str(i+1)].ForceFcn = Fcn
            except:
                pass
            try:
                Fcn = lambda theta, geo : symm_scroll_geo.C2_forces(theta, i+1, geo)
                self.CVs['c2.' + str(i+1)].ForceFcn = Fcn
            except:
                pass
```
Doing this way, `ForceFcn `does not 'freeze' the value of parameter `i`. In the end, all the `ForceFcn` functions are equal to `lambda theta, geo: symm_scroll_geo.C1_forces(theta, 10, geo)`. See [https://stackoverflow.com/questions/3431676/creating-functions-in-a-loop](url) for more explanation.
To fix it, one solution is to use `partial` from module functools. See [https://stackoverflow.com/questions/55396224/how-to-create-a-list-of-functions-using-one-function-with-different-arguments](url).

ibell, 2019-04-12:
Thanks for the bug report.  This is a pretty serious one, I have no idea how it survived before.  My commit (https://github.com/ibell/pdsim/commit/cba2a5f738858208c94eda1c25926d8ba920f108) should have fixed it I think

ibell, 2019-04-15:
Please do let me know if this fixes the problem for you

sebdenis, 2019-04-16:
It's ok for me. Bug fixed!

ibell, 2019-04-16:
Great!

sebdenis, 2019-04-25:
Actually, I have an issue. With a fresh install of PDSim, I retried it and got the followed error :

``
no forces for c1.1 with error: C2_forces() takes at least 3 positional arguments (2 given)
``
and something similar with c1.2, c2.1, c2.2. I think the problem is that you want to fix the parameter ``alpha``, which is not the last one of functions ``symm_scroll_geo.C1_forces`` and ``symm_scroll_geo.C2_forces``.
I solved the problem using the code below:
``
  Fcn = lambda theta, geo, alpha: symm_scroll_geo.C1_forces(theta, alpha, geo)

  self.CVs['c1.' + str(i+1)].ForceFcn = functools.partial(Fcn, alpha=i+1)
``
and the same for c2.

ibell, 2019-05-25:
I don't think that fix works in general, or at least not in Python 3:
``` python
import functools

def f(a1,a2,a3):
    print(a1,a2,a3)    

ff = functools.partial(f, a2=3)
ff(4,5)
```
yields
```
TypeError                                 Traceback (most recent call last)
<ipython-input-1-2c64f0f6e828> in <module>
      4 import functools
      5 ff = functools.partial(f, a2=3)
----> 6 ff(4,5)

TypeError: f() got multiple values for argument 'a2'
```

ibell, 2019-05-25:
But this works:
``` python
import functools

def f(a1,a2,a3):
    print(a1,a2,a3)
    
def ff(a1,a3,a2):
    f(a1,a2,a3)

fff = functools.partial(ff, a2=3)
fff(4,5)
```

ibell, 2019-05-25:
I think this is fixed now

sebdenis, 2019-06-19:
Tested and approved in Python 2. Problem solved for me.

closed 2019-06-20 by ibell