ibell/pdsim #33: Potential discontinuity of scroll_involute_axial_force

sebdenis, 2019-04-12:
Hi,

the axial force on top of the wrap defined by `scroll_involute_axial_force` in scroll.core may not be continue. It comes from a potential discontinuity between sealing points of pockets c2 and d2. 

From the code, for last pocket c2, coordinates of last sealing points are : 
<a href="https://www.codecogs.com/eqnedit.php?latex=\begin{align*}&space;\phi_1&space;&=&space;\phi_{oie}&space;-&space;\theta&space;-2\pi.Ncmax&space;\\&space;\phi_2&space;&=&space;\phi_{oie}&space;-&space;\theta&space;-2\pi.(Ncmax-1)&space;\end{align*}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\begin{align*}&space;\phi_1&space;&=&space;\phi_{oie}&space;-&space;\theta&space;-2\pi.Ncmax&space;\\&space;\phi_2&space;&=&space;\phi_{oie}&space;-&space;\theta&space;-2\pi.(Ncmax-1)&space;\end{align*}" title="\begin{align*} \phi_1 &= \phi_{oie} - \theta -2\pi.Ncmax \\ \phi_2 &= \phi_{oie} - \theta -2\pi.(Ncmax-1) \end{align*}" /></a>
And for pocked d2 :
<a href="https://www.codecogs.com/eqnedit.php?latex=\begin{align*}&space;\phi_1&space;&=&space;\phi_{ois}&space;\\&space;\phi_2&space;&=&space;\phi_{oie}-\theta-2\pi.Ncmax&space;\end{align*}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\begin{align*}&space;\phi_1&space;&=&space;\phi_{ois}&space;\\&space;\phi_2&space;&=&space;\phi_{oie}-\theta-2\pi.Ncmax&space;\end{align*}" title="\begin{align*} \phi_1 &= \phi_{ois} \\ \phi_2 &= \phi_{oie}-\theta-2\pi.Ncmax \end{align*}" /></a>

Transition between pockets c2 and d2 should be done when <a href="https://www.codecogs.com/eqnedit.php?latex=\theta&space;=&space;\theta_d&space;=&space;\phi_{fie}-\phi_{oos}-\pi-2\pi.Ncmax" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\theta&space;=&space;\theta_d&space;=&space;\phi_{fie}-\phi_{oos}-\pi-2\pi.Ncmax" title="\theta = \theta_d = \phi_{fie}-\phi_{oos}-\pi-2\pi.Ncmax" /></a>

To ensure it, the sealing points of pocket d2 should be :
<a href="https://www.codecogs.com/eqnedit.php?latex=\phi_1&space;=&space;\phi_{oos}&space;&plus;&space;\pi" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\phi_1&space;=&space;\phi_{oos}&space;&plus;&space;\pi" title="\phi_1 = \phi_{oos} + \pi" /></a>

ibell, 2019-05-27:
Yes, that analysis seems correct.  I'll fix the code.

ibell, 2019-05-27:
Thanks for the note about this bug.

sebdenis, 2019-06-21:
The problem of discontinuity still remains due to a non perfect transition between pockets 'c' and 'd' around theta_d (don't know exactly why...) It can be fixed using the same trick than in method 'D1_forces' and 'D2_forces' in symm_scroll_geo.pyx:
```
if abs(theta-theta_d(geo))<1e-8: 
     return C1_forces(theta,nC_Max(geo),geo,poly)
```

ibell, 2019-07-04:
I think this is fixed now according to your recommendation, but there still
seems to be a slight discontinuity in sum of the axial forces, which is
perplexing.

closed 2020-04-28 by sebdenis