ibell/pdsim #40: Chamber pairs of radial leakage do not change after the discharge angle

EdwardWooCN, 2020-03-26:
I want to know the radial leakage situation after the discharge angle, but I don't know which part of the code I need to change.
I ran "scroll_compressor.py" under file "examples", and opened the h5 file to check the radial leakage situation. Here's the thing. The scroll compressor I set had only 1 compression chamber pair at maximum, and everything before the discharge angle (theta_d) is fine. Main radial leakage chamber pairs before theta_d are (1) suction chamber--suction channel (s1/s2--sa) , (2) compression chamber--suction chamber (c1/c2--s2/s1). 
But after theta_d, it should be (1) suction chamber--suction channel (s1/s2--sa) , (2) discharge chamber--suction chamber (d1/d2--s2/s1). And I chased back to the "core.py" under file "scroll", and the code related to how to assign radial leakage chamber pairs has no problem. And I can't figure out how the main function works. Would you like to give me some help?

ibell, 2020-03-27:
Can you please show your code for what you did? I am having a hard time to understand what precisely your question is.

EdwardWooCN, 2020-03-27:
[20200327-Question.pptx](https://github.com/ibell/pdsim/files/4394226/20200327-Question.pptx)
Here is the ppt to show what I have done.  My questions are also listed in it.

EdwardWooCN, 2020-03-27:
[examples.zip](https://github.com/ibell/pdsim/files/4394516/examples.zip)
Code file and h5 file.

ibell, 2020-03-31:
Interesting...  I couldn't make out from your picture in the PPT whether the very high radial flow rate was the total radial flow rate or the integrated one over the entire rotation.  It would indeed be surprising if the leakage flow was as high as you mention, but instantaneously it certainly could be.

Getting these chambers right for the leakage paths is something that I've been working on for many years, and I thought that I had it sorted finally, but perhaps not.

EdwardWooCN, 2020-03-31:
Thank you! Your work is really amazing!  The radial flow rate is the average value, which is the integrated one over the entire rotation over 2 PI.