ibell/pdsim #43: silent crash when running the example recip_compressor.py

gfsReboucas, 2020-04-09:
After installing it and running `simple_example.py` successfully I tried to run `recip_compressor.py` but couldn't find the .h5 file it's supposed to generate nor an error message.

I started debugging it until line 1651 of pdsim\core\core.py. VSCode's debug console suddenly closes after I try to run `valve.derivs(self)`. It just leaves this message: `Server[1] disconnected unexpectedly`.

I couldn't go further debugging the .pyx code because I don't know how yet (never used cython). I'm solving it with `solve` (recommended), instead of `precond_solve` (on the original example).

davideziviani, 2020-04-09:
@gfsReboucas : I had some issues with encoding str/bytes for the valve model inside `flow_models.pyx` and I created a `flow_models.py` file instead and it worked. However, I did not figure out what was the issues. The code was silently crashing as you mentioned.

ibell, 2020-04-09:
Is this happening also with the most up-to-date version of the code?  What version of Python do you use?  I made some "fixes" to the code to deal with string encoding, and I fear I didn't make everything better in all ways.

ibell, 2020-04-09:
But recip_compressor.py runs fine on my computer with the master (linux, Python 3.7, CoolProp 6.3.1dev).

davideziviani, 2020-04-09:
@ibell I have not run the latest version of PDSim available here, just the one I have locally and that I do development on, so it can be that encoding or other minor issues have been fixed

ibell, 2020-04-09:
Yes, please try on the master, I believe it might work properly now.

gfsReboucas, 2020-04-10:
@ibell I'm using Python 3.7 on Windows (via Anaconda) and CoolProp 6.3.0. I updated the python version on conda_environment.yml (2.7 to 3.7) and used it to create an environment for pdsim.

I want to see what happens when @davideziviani try it using the master repo. I think it won't work.

Thank you both for the quick reply.

ibell, 2020-04-11:
Wel... there was a bytes/str issue, but I just merged the discvalve branch and that should be fixed.

ibell, 2020-04-11:
Please try with the current master, on OS X and in linux the ``recip_compressor.py`` file runs.

gfsReboucas, 2020-04-11:
It works now, thanks!

closed 2020-04-11 by gfsReboucas

davideziviani, 2020-04-11:
Worked for me as well!

ibell, 2020-04-11:
It's really weird that you got a crash instead of an exception.  But
nonetheless, glad it's fixed.