ibell/pdsim #47: Valve can exceed their limits

ibell, 2020-04-12:
Valves can be more than closed, or exceed the valve stop.  I'm not sure where best to integrate the fix for this, basically we need to threshold the valve to be in [0,x_stop], but some thinking needs to go into where that constraint is implemented.

ibell, 2020-04-12:
@davideziviani what do you think?  I think this has been the cause of some weird valve stuff.

davideziviani, 2020-04-12:
I am running rolling piston compressor model now and I will get back to you as soon as the simulation finishes. The mass flow rate balance was very accurate, now I am waiting to complete the run

ibell, 2020-04-12:
Thanks, I think this might need a bit of discussion to get perfect.  It was pretty tricky to figure out how to fix the consistency of the valves...

davideziviani, 2020-04-12:
Mass flow rate balance is ~3% and it used to be >0.1%. I need to do some thinking and look at some other coding that I have done for checking valve lift profile. Also, I think I would like to add an additional debug plot that allows cycle-to-cycle history plotting

ibell, 2020-04-12:
Weirdly, the mass balance is still fine on scrolls, but not on all machines
(e.g., piston expander with Euler is fine, but not with RK45), and it is a
bit worrying.  I wonder if it is something about our method of integration
to calculate the average mass flow rate?

davideziviani, 2020-04-12:
I am looking exactly at that. I have another code written in MATLAB specifically for recip compressors to ensure situations such as a valve stays open at end of the 2pi that obviously needs to be coherent with the beginning of the cycle to ensure continuity. Integration is done with RK45 (ODEs + valves) and there is a modified Broyden method external to ensure OEB of the lumps and rotational speed. I will share the code with you. I am now running Z-compressor