ibell/pdsim #56: Modeling mixtures in scroll compressor

mohsint7, 2020-09-18:
This issue is related to both PDSim and Coolprop.

I am trying to model mixtures (R454B.mix) using existing PDSim scroll core class. I am getting some errors like "DHSU_T_flash does not exist" or need to build the phase envelop. 

The complete core class is structured around CoolProp  State.update or State.copy. Due to some missing functions for mixtures in Coolprop, I got these errors. Is there any way around these errors?

I also tried calling REFPROP using Coolprop but that is extremely slow and has convergence issues if, during iterations, temperature is in near critical region.

davideziviani, 2020-09-18:
@mohsint7: these are well known issues. I am working with R454C.mix with REFPROP10, and by using REFPROP backend it works. It is slow due to the wrapping around REFPROP. Are you intentionally running nearby C.P. ?

mohsint7, 2020-09-18:
Thank you for quick response.
I am not intended to run it for near C.P but trying different lump temperature guess values to avoid Pr number error in PDSim code.  during iteration, it automatically go near C.P.

using REFPROP, a single iteration is taking hours. is it normal?

davideziviani, 2020-09-18:
@mohsint7  nope, it takes just minutes to run. There must be something else for instance geometric parameters.

Inside core/core.py , in the OBJECTIVE_CYCLE( )  you can add a check for Tlumps to avoid unrealistic values.

For multi-lumps:
                    # Check the new Tlumps
                    for jj in range(len(Tnew)):
                        if Tnew[jj] > 400 or Tnew[jj] < 273:
                            Tnew[jj] = 340 #[K]

ibell, 2020-09-19:
Also, please see the example ``examples/scroll_compressor_bicubic_with_mixtures.py`` which shows how to use tabular interpolation with mixtures.  That can help with speed.  You might also want to provide a better guess value for the lump temperature(s).