ibell/pdsim #57: time.clock deprecated since py3.3 removed py3.8; pdsim incompatible

danieldjewell, 2020-09-24:
According to [the Python3.7 documentation on the time module](https://docs.python.org/3.7/library/time.html#time.clock), time.clock() has been deprecated since py3.3 and was indeed removed in py3.8 thus breaking pdsim at least here (and in other places): 

https://github.com/ibell/pdsim/blob/d957202a53503508442aa5a325091482eb1f13bd/PDSim/core/core.py#L5

I recognize that at least in the conda env file, python 3.7 is targeted -- but since we're on the cusp of 3.9 being final (looks like 3.9.0rc2 was released on 2020-09-24) and 3.8 is reasonably mainstream, I thought it might be useful to document this. 

I haven't gone through the code yet to see where/how ```clock()``` is being used but it seems at first glance that ```time.process_time()``` might be a drop-in replacement.

davideziviani, 2020-09-24:
@danieldjewell : we are aware of that issue and we have a few other issues with py3.8 . A PR is welcome.

ibell, 2020-09-24:
timeit.default_timer is my go-to for timing.  Uses most accurate timer
available

ibell, 2020-10-27:
This is a fix for the GUI part at least: https://github.com/ibell/pdsim/commit/025319e573fc38d969b54811c2291e2b31059456 .  Same solution for core/core.py

commit 95f8c7a2 (2020-11-17): Two more time.clock->timeit.default_timer

Closes #57
closed 2020-11-17 by ibell