ibell/pdsim #58: Using PDSim in the cloud with Jupyter Notebook/Google Colab

MaEnr, 2020-10-02:
I am trying to run a compressor model that uses both PDSim and CoolProp in either Google Colab or Jupyter Notebook through the cloud. I am trying to do this through the cloud because one of the goals for my project is to run models without downloading anything locally.

I was able to install PDSim using the setup.py file inside Google Colab and PDSim version 2.13 shows up under the list of packages in the notebook. However, I am unable to import from PDSim.flow.flow and PDSim.core. I did not use msvc to build and install the package so I am wondering if this could be my issue. Is there any way around using something like msvc?

ibell, 2020-10-06:
I don't think the compiler will be the issue, as your image is Linux
based?  Or not?

Can you share your image?

Does import PDSim work?

MaEnr, 2020-10-09:
I'm sorry I do not know what you mean by image. 

Running python setup.py results in creating and installing a .egg file. After that, import PDSim seems to be working however import PDSim.core results in this error: ModuleNotFoundError: No module named 'PDSim.flow.flow_models'. 

I have attached a screenshot showing the imports with the error, as well as a screenshot of some of the output for running !python setup.py install. 

[image]
[image]

ibell, 2020-10-10:
Once you have installed pdsim with python setup.py install, you don't need
to add the location to sys.path (that's the whole point of the install).
Maybe not putting in the sys.path modification will help?  You'll note that
process is not done in any of the PDSim examples.