ibell/pdsim #59: how to use GUI

EdwardWooCN, 2020-10-15:
I have some doubts in how to use GUI. Here are my installation steps.
Step1: Installing PDSim env in conda: [Successfully]
`conda env create -n pdsim_stable -f RTDenvironment.yml`
[image]

Step 2: Switch the path under PDSim-master folder and run setup.py [Successfully]
[image]

Step 3: Switch to the folder .\GUI and run setup.py [Error]
[image]
`running build
running build_exe
Traceback (most recent call last):
  File "setup.py", line 113, in <module>
    pack(get_defaults())
  File "setup.py", line 85, in pack
    executables = [GUI2Exe_Target_1]
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\site-packages\cx_Freeze\dist.py", line 342, in setup
    distutils.core.setup(**attrs)
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\distutils\core.py", line 148, in setup
    dist.run_commands()
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\distutils\dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\distutils\command\build.py", line 135, in run
    self.run_command(cmd_name)
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\site-packages\cx_Freeze\dist.py", line 217, in run
    freezer.Freeze()
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\site-packages\cx_Freeze\freezer.py", line 638, in Freeze
    self.finder = self._GetModuleFinder()
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\site-packages\cx_Freeze\freezer.py", line 377, in _GetModuleFinder
    finder.IncludeModule(name)
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\site-packages\cx_Freeze\finder.py", line 653, in IncludeModule
    namespace = namespace)
  File "C:\Users\NH55\anaconda3\envs\pdsim_stable\lib\site-packages\cx_Freeze\finder.py", line 349, in _ImportModule
    raise ImportError("No module named %r" % name)
ImportError: No module named 'numpy.core._multiarray_tests.cp37-win_amd64'`
cx_Freeze version is cx_Freeze-6.2-cp37-cp37m-win_amd64.whl
numpy version is 1.19.11
glob2 version is  0.7

ibell, 2020-10-16:
That ``setup.py`` in the GUI folder is for packing the GUI into an executable package, whereas what you want to do is just do "python PDSimGUI.py" to launch the GUI.  On some systems you might want to use ``pythonw`` instead of ``python``.

EdwardWooCN, 2020-10-16:
It didn't work also.
[image]

ibell, 2020-10-16:
You want to run the command starting in the GUI folder.

ibell, 2020-10-27:
If run from GUI folder, it will work.

closed 2020-11-17 by ibell