ibell/pdsim #68: Install Process Guidance

huges84, 2021-02-04:
Hello,
First off, let me start by saying I am a mechanical engineer and good with computers but very novice with programming and python, so hopefully my feedback is useful and not too elementary. I had some difficulties getting up and running following the Read The Docs information, so I wanted to give my feedback. BTW, I was doing all of this in Windows 10 x64 with admin rights.

If I understand correctly, conda is used instead of pip to avoid compiling, handle version conflicts and create environments. However, I ran into errors using it that I think others will as well.
 
1. When trying to run the command "conda install matplotlib numpy scipy h5py cython pip wxpython pyyaml" it would fail to find a working combination of those packages, plus the latest Python (3.8) installed with Miniconda. 
2. Because of that failure, with mountains of error text, I tried installing each package by itself, using conda. That worked for every package except wxpython, which would not install due to the python version being too high for the wxpython package available in conda. When using "conda search wxpython" it only sees versions up to version 4.0.4, which is older. 
3. When going to the wxpython site to understand how to get the right version, they show newer versions of wxwidget (v 4.1.1) that are compatible with python 3.8. They point you to using pip to install. I almost tried to, but then I googled conda vs pip and came to understand that I must use conda to install packages for them to be available in the environments. I did find a conda version at https://anaconda.org/conda-forge/wxpython but I did not use it, due to the below.
4. I also learned that it is important to install all needed conda packages at once, so the installer can check for version conflicts and pick a compatible combination of versions.
5. I found that there is wxpython Phoenix that has revived updates to the project and conda versions are available on conda-forge. I almost installed from there, but then I saw that PDSim conda-enviroment.yml mentions python 3.7. I learned that I could install different versions of python side by side, so I installed python 3.7.
6. It's unclear to me why CoolProp can be installed via pip but the others are via conda. I am guessing it's because coolprop is used only to compile PDSim.
7. I got the warning "warning: PDSim\scroll\symm_scroll_geo.pyx:787:20: Unreachable code" pretty early in the build process. The full shell output from the compile is attached.
8. When trying to run the GUI from shell using "python pdsimgui.py" I get the following error "ModuleNotFoundError: No module named 'quantities'"

So my recommendations to update Read The Docs are:
A. If there is a specific version of Python to use, indicate so in the install instructions and have the user install the Miniconda version that packages the correct python version in the installer.
B. Make clear that the conda install step should not be broken into pieces, instead it should be run in one big install. If there's errors then the user should try an older python version that all packages might be compatible with.
C. Perhaps make use of the conda-forge version of wxpython, because it seems that the normal conda wxpython is not being maintained.
D. I had to do some activation of conda to get it to work in the PowerShell after installing Miniconda. I have forgotten the steps, but some guidance or at least mention of it might be good. 
E. I would add some basic instructions on getting to the point that you can run conda and not have the command line immediately close (open PowerShell by holding shift and right clicking in Miniconda3\Scripts folder).
F. For pulling the repo from github with Git, you might want to mention that all options can be left at defaults (no boxes checked).
G. For the step to compile PDSim, I believe that "python setup.py install" has to be run from the pdsim directory containing the code. If so, that should be mentioned. I opened my PowerShell window in that directory before running the command and it worked.
H. The step after step 8 is labeled as 1 and it says "Now go into the doc folder, start IPython, .." This is the first mention of IPython. At first I thought it was a typo for Python, but I see it's not. Installing IPython or maybe Jupiter should be mentioned first.
I. The Linux instructions on RTD also mentions Python 2.X. 
J. The instructions for creating an environment and running PDSim should be much higher on the page, IMO. I would put them right after the Linux instructions. I missed them a few times.

The "pdsim\doc\tutorial" folder has really outdated information, referring to Python 2.x and VS 2008. It's probably best to delete this entire folder and just point to the RTD page.
[pdsim_compile_log_2021-02-04-14_34.txt](https://github.com/ibell/pdsim/files/5928360/pdsim_compile_log_2021-02-04-14_34.txt)
[pdsimgui_error_2021-02-04-15_31.txt](https://github.com/ibell/pdsim/files/5928361/pdsimgui_error_2021-02-04-15_31.txt)

huges84, 2021-02-04:
To resolve the issue running the GUI, I ran the setup.py file in pdsim directory using powershell. (There was no instructions to do so, I just found that file made reference to quantities folder and gave it a shot). That didn't seem to work, because I still got errors about the quantities module. 

Then I installed the quantities package using "conda install -c conda-forge quantities'. That then allowed pdsimgui.py to launch without error. I also ran the default inputs and solver without issue.

davideziviani, 2021-02-04:
I use Py3.8 through Anaconda and MSVC2019 for compiling and everything works well. I would skip Py3.7.

huges84, 2021-02-04:
The main conda repo does not have a version of wxpython compatible with 3.8. Also the latest release of PDSim (v2.13) states "Hold off on python 3.8 for now". However, I think that by following the directions I am using the latest GitHub code. Would you recommend that I use the conde-forge version of wxpython and python 3.8 with the latest PDSim code?

davideziviani, 2021-02-04:
Not true anymore. wxpython 4.1.1 works like a charm. There could be a few minor syntax issues but mostly they have been taken care of. I am running Py3.8.5 and I have no issues.

ibell, 2021-02-04:
@huges84 I too am a mechanical engineer !  So is @davideziviani !

I pushed some changes to the docs, hopefully they help to address some of your issues you ran into.  

Regarding your questions: the pip package for CoolProp is always up to date, not always the case with the conda package (I think).  So always better to use that.

ibell, 2021-02-04:
Yeah, the tutorial is indeed out of date.  Once you get a bit up to speed with PDSim, want to write one?  You clearly have the right mindset to contribute that, and it is better if it is written by a non-expert because then all the things that are not obvious to you get documented.

huges84, 2021-02-05:
I am happy to contribute however I can. I will need to poke around a bit first. 

What is the best way to contact you for questions? I will try to respect your time by researching python questions before asking those type of questions and looking at the code, papers and thesis before asking PDSim questions.

michox, 2021-08-14:
If anybody else is having issues with Conda make sure that you don't have another python installed and make conda your default python. that worked for me.