ibell/pdsim #69: StateFlooded TypeError

gapablo, 2021-02-12:
Hi dev team,

I'm trying to run "Example_StateFlooded.py" and receiving the following error:
```
Traceback (most recent call last):
  File "/(...)/pdsim-master/test/Example_StateFlooded.py", line 3, in <module>
    S=StateFlooded("R245fa", 'POE', 1200, 373.15, 0.1, 'HEM')
  File "PDSim/core/state_flooded.pyx", line 43, in PDSim.core.state_flooded.StateFlooded.__init__
  File "stringsource", line 15, in string.from_py.__pyx_convert_string_from_py_std__in_string
TypeError: expected bytes, str found
```
suggesting a compatibility problem with Python 2 and 3.

When trying an simple workaround and calling StateFlooded as S=StateFlooded(b"R245fa", b'POE', 1200, 373.15, 0.1, b'HEM'), the following error occurs:
```
Traceback (most recent call last):
  File "/(...)/pdsim-master/test/Example_StateFlooded.py", line 3, in <module>
    S=StateFlooded(b"R245fa", b'POE', 1200, 373.15, 0.1, b'HEM')
  File "PDSim/core/state_flooded.pyx", line 56, in PDSim.core.state_flooded.StateFlooded.__init__
  File "CoolProp/CoolProp.pyx", line 721, in CoolProp.CoolProp.State.set_Fluid
  File "stringsource", line 38, in string.to_py.__pyx_convert_PyUnicode_string_to_py_std__in_string
MemoryError
```
Do you have a working StateFlooded version, or can you help me to fix this issue?

I followed your instructions to install PDSim and  included "PDSim/core/state_flooded.pyx" in the compiler list in "setup.py". I'm using Python 3.8.6 and coolprop 6.4.1 in the conda environment as suggested by you.

Your support is very much appreciated.

davideziviani, 2021-02-12:
Hello! this is a well known issues between bytes and strings. We are working on fixing this issues, but I do not have bandwidth for a PR. Feel free to create a PR.

gapablo, 2021-02-12:
Thank you for the quick response. Do you have a suggestion on how to approach this issue and what parts of the code would need to be changed?

gapablo, 2021-02-16:
Manually reinstalling CoolProp as `conda install conda-forge::coolprop` and initializing the StateFlooded class with byte strings as `S = StateFlooded(b'R245fa', b'POE', 101.325, 300.0, 0.01, b'HEM')` seems to have fixed the issue for me.

closed 2021-02-16 by gapablo