ibell/pdsim #72: Screw or scroll expander

David-Mamrilla, 2022-02-17:
Hello,
I have installed PDSim and am very impressed with the simulation interface for scroll and piston expanders. I would like to kindly ask, does anyone posses models for a scroll or screw expander? I saw that there is a model for a piston expander in the "examples" folder, but I was wondering whether it is possible to implement it with the GUI interface?

Thank you.

ibell, 2022-02-20:
Did you check out the models from Paper II?

Yes, it could be implemented, if you wanted to do the work to do so, or if
you had funding for me to do it.

David-Mamrilla, 2022-02-21:
Hello, thank you for replying,

Yes, I did check both of the papers. and read carefully especially the part about screw expanders in paper II. However, would it possible to find and use the actual python model? My intent would be to use it for academic purposes only, for my Master thesis on possibilities of PD expanders in steam cycles.

Thank you.

ibell, 2022-02-21:
The files are here, but seem to be embargoed:
https://figshare.com/articles/code/PDSim_source/7985651.  I'll investigate.

ibell, 2022-02-21:
Embargo lifted (should have been lifted a long time ago).

> Message ID: ***@***.***>
>

David-Mamrilla, 2022-02-22:
Thank you for unlocking the files. Maybe I am not looking in the right folders but I cannot see a screw expander model? It seems that the files on figshare are the same ones as here on Github, with examples of scroll and reciprocating compressors and a piston expander, but no screw.
Thank you and sorry if it is in one of the folders and I just didn't see it.

ibell, 2022-02-23:
@davideziviani Any idea where the examples went to? I agree they don't seem to be in the figshare repo from paper I

David-Mamrilla, 2022-03-08:
Hello, 
have you by any chance managed to find the examples?

MAKlNE, 2022-04-02:
Hello,
I can only use Recip compressor and scroll compressor in PDSim app.

Is the Scroll Expander locked?

Kind Regards.
Alan

David-Mamrilla, 2022-04-03:
No, it is not locked. Only recip and scroll compressor have been implemented in the app. However, there should be python models for other machines, among them the scroll and screw expanders. But it seems the authors can't find them at the moment.