ibell/pdsim #74: Is the specified mechanical efficiency equation wrong ?

caskandavor, 2023-01-03:
First of all, thank you very much for your work, I have been working on scroll compressors for a few months now and PDSim has been very helpful. 

I tried to get into the source code the understand the model better. I think I may have found a mistake. I believe that the mechanical losses are wrong when selecting specified mechanical efficiency. I enclosed two screenshots where I explain why I believe there has been a mistake. 

If you still believe the equation used in the code is correct, I would very much appreciate an explanation.

Thank you again and happy new year !  

[image]
[image]

ibell, 2023-01-04:
I think the formula is indeed wrong, but not in the way you write.

First, let's agree about how to define the mechanical efficiency. I would say it is the indicated power divided by the total mechanical power which is the indicated power plus the mechanical losses.  Thus:

$$\eta_{me} = \frac{W_{pv}}{W_{pv}+W_{mech}}$$

and solving for mechanical losses, you have

$$ W_{mech} = W_{pv}*(1/\eta_{mech}-1)$$

ibell, 2023-01-04:
What do you think?

davideziviani, 2023-01-04:
Actually in our development version at Herrick we fixed this issue, but forgot to commit

caskandavor, 2023-01-04:
@ibell  Thank you for your answer. I agree with what you wrote and it's actually how I corrected it in the code (since $W_{mech}$  is only computed afterwards we indeed need to use $W_{pv}$ ).

There is however something else that I believe might be erroneous. It is the way `Wdot` (Power of Compressor) is defined:

[image]

Here it is said that `self.Qamb` is **positive** if heat is being **added to the lumped mass**. If we consider this equation to be a conservation equation **on the refrigerant**, it makes sense (if heat is lost to ambient, `self.Qamb` is negative so we need to provide this work in addition to the enthalpy change of the gas). However this is not consistent with the way `self.Qamb` is defined: 

[image]
[image]

The heat lost to the ambient is modeled with Newton's law of cooling $Q_{amb}=h_{shell}\cdot A_{shell}\cdot(T_{shell}-T_{amb})$, which will be positive if heat is being **lost from the lumped mass**.

When the residuals for the overall energy balanced are computed, the sign of `Qamb` is flipped, and so is the convention for heat transfer signs (`Qamb` is positive if heat is being removed)

[image]

For this reason, I tried to flip the sign of `self.Qamb` in the first screenshot equation and align myself with the last convention (`Qamb` is positive when heat is being removed). I did a quick study comparing the before and after this modification in the code. Both cases have the exact same parameters, each plot is obtained by running over 3 increasing values of **HTC** (the heat transfer coefficient inside the scroll compressor), but `h_shell` is kept constant (please not that the values of HTC on the x-axis are not the values, just the number of the run).

[image]

As you can see, with the un-modified equation, the higher the heat transfer coefficient, the **lower** Power of compressor `Wdot` is, which does not seem resonable. Also, while the evolution of the electrical power and the gas power have the same evolution, the "power of compressor" has a different one, which seems odd to me.

With the modified equation, the latter effect is removed, and the curves now follow the same trend. However there still is the problem that the power consumption decreases as the HTC increases and the heat losses increases coincidentally. This does not make physical sense to me. I expect the power consumption to **increase** as the heat lost to the ambient **increases** as we have to compensate for it.

What do you think about this ?

Thanks again and sorry for the long comment.

xinding1207, 2023-01-05:
My thought is that the you may mis-interpret the meaning of variables. "Wdot" should be the energy received by the refrigerant from the compressor instead of "power of the compressor" and does not involved in calculations of the total power input of the compressor. Please notice that in both figures (modified and un-modified), the Wdot_mechanical and Wdot_pv were not affected by the Wdot value at all. If the heat transfer increases while the total power input is constant, the energy received by the refrigerant will decrease, which explains the trend you observed in the left sub-figure. Therefore, judging the sign of Qamb in different equations may be misleading.

caskandavor, 2023-01-05:
@xinding1207 Agreed, `Wdot` is indeed not involved in the calculation of the total power input (hence the fact that despite in changing from one figure to the other, the `Wdot_electrical` and `Wdot_pv` remain the same.  I also agree that the sign of `Qamb` can be different depending on the equation and the control volume studied and that it could lead to errors.
However, the sub left figure is obtained with the code un-modified, and it shows that the power of the compressor `Wdot` (which I agree, given its definition, is the energy received by the refrigerant from the compressor) **decreases** as the heat transfer increases. Note that the heat transfer to the ambient is positive, and from the convention in the code, it means that heat is **lost**. If heat is **lost**, the power given to the refrigerant should **increase** to compensate the heat loss.

As you can see in the previous comment, in the code, different conventions are used in different parts of the code for `Qamb`.
I simply believe that the one used to compute `Wdot` is not correct.

Also, since we've established that `Wdot` has no influence on the total power input `Wdot_electrical` and `Wdot_pv`, let's forget about it altogether for my next point. We can see that `Wdot_electrical` and `Wdot_pv` **also decrease** as the heat transfer with the ambient increases and heat is being lost. That should be the other way around for the reasons stated before.

I am still trying to figure that out, I will keep you updated. Please do comment if you have an idea or see a flaw in my reasoning !

caskandavor, 2023-01-05:
One thing that I had not taken into account is the parameter `suction_fraction`, which is the **fraction of the motor losses going in the suction gas**. For now, it was set to 1, so all the losses from the motor were heating the gas. Redrawing a sankey diagram: 

[image]

The way I see things, `Wdot` ( $W_{comp}$ ) is just another way to write the total power input (so $W_{el}$ ) since `Qamb` represents all the losses anyway!