ibell/pdsim #75: ValueError for certain cases of PDSim

caskandavor, 2023-03-09:
Hi all,

I have been using PDSim to better understand scroll compressors, but something has been bugging me for a while. For some cases (typically for high discharge pressures), there is a `ValueError` (`options.p is not valid in saturation_T_pure_1D_P`). As seen in the screenshot below, this seems to come from CoolProp, as we try to get a pressure from a non valid temperature (here T = -303.394 K).

Note that this happens at the very beginning of the run (not even the first computing iteration is completed).

Also, I sometimes get around this by reducing the flank and radial gaps to a minimum, but even the smallest gaps do not work for too high of a pressure (here 15 [bar] of gauge pressure).

Have you encountered such cases as well ?

Thank you!

[image]

ibell, 2023-03-09:
Can you share an example where this failure happens?

caskandavor, 2023-03-09:
Do you mean all the parameters that I used? If yes, here is an example which generates an error: (Note all solver parameters are lest as default).

# Geometry
## Scroll wrap inputs
Displacement $[m^3/rev]: 0.00007643$
Volume ratio $[-]: 2.67$
Thickness of the scroll wrap $[m]$: 0.0045
Orbiting radius $[m]$: 0.002387
$\phi_{i0}$ $[rad]: 2$
$\phi_{is}$ $[rad]: 2$
$\phi_{os}$ $[rad]: 2$
Flank gap width $[m]: 10e-6$
Radial gap width $[m]: 10e-6$

## Discharge region inputs

2 Arcs with Perfect Matching Profile 
Discharge port diameter $[m]: 0.015$

## Tube inputs

Inlet tube length $[m]: 0.0$
Inlet tube inner diameter $[m]: 0.02$
Outlet tube length $[m]: 0.0$
Outlet tube inner diameter $[m]: 0.02$

# States
## Rotational speed 
Rotational speed $[rad/s]: 209$
## Inlet State
Fluid: Air
Temperature $[K]: 293.15$
Pressure $[kPa]: 101.325$
## Discharge State
Discharge pressure $[kPa: 1601.325]$

# Flow model parameters
Left as default (All $X_d=0.8$)

# Mechanical
## Motor Model
Motor efficiency $[-]: 0.95$
## Heat transfer inputs
Shell-ambient mean HTC $[kW/m^2/K]: 1.0$
Shell outer area $[m^2]: 0.0405$
Ambient temperature $[K]: 333.15 K$
HTC in the scrolls $[kW/m^2/K]: 0.2$
Fraction of motor losses: $0$
## Orbiting Scroll Inputs
Left as default
## Mechanical loss Models
Specified mechancial efficiency $[-]: 0.9$

caskandavor, 2023-03-10:
Oh right here it is: 
[example.txt](https://github.com/ibell/pdsim/files/10939699/example.txt)

caskandavor, 2023-03-14:
@ibell was this the generated script you were refering to ?

ibell, 2023-03-15:
How old is the GUI you are using, it is generating Python 2.7 code.
Everything is in python 3, and I can't even get your script to start
running :/

caskandavor, 2023-03-16:
I have generated this script with the 2.9 version of the GUI.
For my work, I use PDSim 2.14 but without the GUI.