ibell/pdsim #79: PDSim GUI failed to use REFPROP fluid

Wang-Zongwei, 2023-08-18:
Today I installed pdsim on a new PC (windows 11) with latest github code. I found below issue, and I hope my feedback is useful and not too elementary. 

1. Installation process is very smooth except below:
	ModuleNotFoundError: No module named 'pandas'

	This issue is easy to solve, only need input "pip install pandas", then it's OK.

2. PDSim GUI failed to use REFPROP fluid:
	The lastest version (PDSim 2.14 and CoolProp 6.5.0) cannot call REFPROP fluid in GUI. But the 2.9H windows setup version (downloaded from https://sourceforge.net/projects/pdsim/files/) seems OK to call the REFPROP fluid in GUI. Attached is related picture. (1st pic is based on PDSim GUI 2.14; 2nd pic is based on PDSim 2.9)

	Please help check if this issue could be fixed? Thanks very much.

[image]
[image]

ibell, 2023-08-20:
How about REFPROP use without Bicubic interpolation, is that still ok?

Wang-Zongwei, 2023-08-23:
> How about REFPROP use without Bicubic interpolation, is that still ok?

ibell, 2023-08-23:
What error do you get?

Wang-Zongwei, 2023-08-24:
For 2.14 conda GUI version, with o.r without Bicubic both NOK to call REFPROP. There is no any response if I select the "+" fluid:
BICUBIC&REFPROP::R32[0.697614699375863]&R125[0.302385300624138] or
REFPROP::R32[0.697614699375863]&R125[0.302385300624138] or
[image]
[image]

ibell, 2023-08-25:
Fixed, there was an error message in the main output log

Wang-Zongwei, 2023-08-25:
Thanks so much.

Wang-Zongwei, 2023-08-26:
I found latest verison from "/pdsim/tree/HEOSGUI" already solved this issue. Thanks so much!!!

closed 2023-08-26 by Wang-Zongwei