ibell/pdsim #80: Different discharge port diameter but exactly same result

Wang-Zongwei, 2023-08-18:
I want to study with PDSim GUI the impact of different discharge port diameter on scroll compressor performance. But I found it seems no performance change (volumetric and isentropic efficiency) under different discharge port diameter input.

GUI version: PDSim 2.14 and CoolProp 6.5.0

Do you think such result is reasonable? Thanks very much.
[image]

ibell, 2023-08-19:
Are the coordinates of the port the same? You might want to put in the port
coordinates yourself

Wang-Zongwei, 2023-08-26:
I did't input any coordinates for the discharge port (all x,y keeps blank). I will try with different coordinates input data to see if the performance result is different.

Wang-Zongwei, 2023-08-26:
I have got reasonable performance result with different coordinates input data for discharge port (see pic).
I will use this method for my future study.
[image]

Wang-Zongwei, 2023-08-26:
Thanks for your help.

closed 2023-08-26 by Wang-Zongwei