ibell/pdsim #82: Problem with starting simulations

tomam28, 2023-08-25:
Hello pdsim community,

I installed all required packages (with the required versions, because newer cpython version have problems with declaring variables with cpdef instead of cdef)  and run setup.py sucessfully. The GUI works also fine, but when I want to solve a simulation, then this error appears: 

`Traceback (most recent call last):
  File "C:\...\pdsim\examples\simple_example.py", line 204, in <module>
    Compressor(solver_method = 'Euler')
  File "C:\...\pdsim\examples\simple_example.py", line 186, in Compressor
    recip.solve(key_inlet='inlet.1',
  File "C:\...\pdsim\PDSim\core\core.py", line 1479, in solve
    self.OBJECTIVE_CYCLE(x0, self.x_state, 
  File "C:\...\pdsim\PDSim\core\core.py", line 1137, in OBJECTIVE_CYCLE
    self.one_cycle(X, 
  File "C:\...\pdsim\PDSim\core\core.py", line 1030, in one_cycle
    aborted = integrator.do_integration(N, tmin, tmax)
  File "C:\...\pdsim\PDSim\core\integrators.py", line 80, in do_integration
    self.xnew = self.xold + self.h*self.f1
TypeError: unsupported operand type(s) for *: 'float' and 'PDSim.misc.datatypes.arraym'`

Thanks for any help! :)

[image]

ibell, 2023-08-25:
Can you post the generated script if it doesn't include anything top secret?

tomam28, 2023-08-28:
Thanks for the fast reply!

The generated script is:
```python

# This file was automatically generated by PDSimGUI on Mo, 28 Aug 2023 07:45:45
# Versions used:
# PDSim: 2.14
# CoolProp: 6.5.0 git revision: bafdea1f39ee873a6bb9833e3a21fe41f90b85e8
# 
# To run this file, in a console, type
#
#     python this_file_name.py
#
# where this_file_name.py is the name of this file
#
# In python 2.7 make a/b always give the double division even 
# if both a and b are integers - thus 2/3 now yields 0.666666666... 
# rather than 0
from __future__ import division

# NO PLUGIN (post_import) 

# General python imports
import time, sys, os, timeit

# Plotting and numeric things
from math import pi
import numpy as np
from matplotlib import pyplot as plt

# Imports from PDSim that are needed for all families of machines
from PDSim.flow.flow import FlowPath
from PDSim.flow.flow_models import IsentropicNozzleWrapper
from PDSim.core.containers import ControlVolume, Tube
from PDSim.core.motor import Motor
from PDSim.plot.plots import debug_plots

# Imports from CoolProp
from CoolProp import State
from CoolProp import CoolProp as CP

# Add the paths for any additional plugin folders (hard-coded absolute paths)
sys.path.extend([])
# NO PLUGIN (post_import) 

def build():
    # NO PLUGIN (pre_build) 
    
    from PDSim.scroll.core import Scroll
    
    # NO PLUGIN (pre_build_instantiation) 
    
    sim = Scroll()
    
    sim.run_index = 6
    sim.description = ""
    # NO PLUGIN (post_build_instantiation) 
    
    #############
    # From GeometryPanel
    ############
    
    
    #  Parameters from the GUI
    Vdisp = 0.0001048 #[m^3/rev]
    Vratio = 2.2 #[-] 
    t = 0.004 #[m]
    ro = 0.005 #[m]
    phi_i0 = 0.0 #[rad]
[... 263 lines cut ...]
    self.xnew1 = self.xold+self.h*(1.0/5.0)*self.f1
TypeError: unsupported operand type(s) for *: 'float' and 'PDSim.misc.datatypes.arraym'
Thread-2: Process is done

Log from thread #2
-------------------

Log from thread #3
-------------------

Log from thread #4
-------------------

Log from thread #5
-------------------

Log from thread #6
-------------------

Log from thread #7
-------------------

``` 

Thanks for the help!

ibell, 2023-08-29:
I ran that script on the development version of PDSim, no errors, ran fine.

tomam28, 2023-09-25:
Its all solved now, there was a problem with my conda version! 👍

closed 2023-09-25 by tomam28