ibell/pdsim #84: PDSim GUI failed to Euler/Heun integrator

Wang-Zongwei, 2023-08-27:
Today I found below issue, and I hope my feedback is useful.

PDSim GUI failed to Euler/Heun integrator (RK45 is OK). All the scroll input data is from the default setting.

Wang-Zongwei, 2023-08-27:
Euler error:

About to run the script file C:\Users\Administrator\.pdsim-temp\script_bdf61f53bc46.py
C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\scroll\core.py:382: UserWarning: xport and yport not provided, defaulting back to circular discharge port; should be stored in self.geo.xvec_disc_port and self.geo.yvec_disc_port
  warnings.warn('xport and yport not provided, defaulting back to circular discharge port; should be stored in self.geo.xvec_disc_port and self.geo.yvec_disc_port')
caching discharge port blockage, please wait... done
Process Run1-1:
Traceback (most recent call last):
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\pdsim-master0826\GUI\processes.py", line 77, in run
    script_module.run(self.sim, pipe_abort = self.pipe_abort)
  File "C:\Users\Administrator\.pdsim-temp\script_bdf61f53bc46.py", line 273, in run
    sim.precond_solve(key_inlet = 'inlet.1',
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 861, in precond_solve
    self.solve(**kwargs)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1479, in solve
    self.OBJECTIVE_CYCLE(x0, self.x_state,
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1137, in OBJECTIVE_CYCLE
    self.one_cycle(X,
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1030, in one_cycle
    aborted = integrator.do_integration(N, tmin, tmax)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\integrators.py", line 74, in do_integration
    self.f1 = self.derivs(self.t0, self.xold)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 109, in derivs
    return self.sim.derivs(t, x)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1632, in derivs
    self.core.properties_and_volumes(self.CVs.exists_CV, theta, STATE_VARS_TM, x)
  File "PDSim\core\containers.pyx", line 295, in PDSim.core.containers.CVArrays.properties_and_volumes
  File "PDSim\core\containers.pyx", line 342, in PDSim.core.containers.CVArrays.properties_and_volumes
  File "CoolProp\CoolProp.pyx", line 781, in CoolProp.CoolProp.State.update_Trho
  File "CoolProp\AbstractState.pyx", line 104, in CoolProp.CoolProp.AbstractState.update
ValueError: p is not a valid number
Thread-2: Process is done
[image]

Wang-Zongwei, 2023-08-27:
Heun error:

About to run the script file C:\Users\Administrator\.pdsim-temp\script_ae3e98385bb1.py
C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\scroll\core.py:382: UserWarning: xport and yport not provided, defaulting back to circular discharge port; should be stored in self.geo.xvec_disc_port and self.geo.yvec_disc_port
  warnings.warn('xport and yport not provided, defaulting back to circular discharge port; should be stored in self.geo.xvec_disc_port and self.geo.yvec_disc_port')
caching discharge port blockage, please wait... done
Process Run1-1:
Traceback (most recent call last):
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\pdsim-master0826\GUI\processes.py", line 77, in run
    script_module.run(self.sim, pipe_abort = self.pipe_abort)
  File "C:\Users\Administrator\.pdsim-temp\script_ae3e98385bb1.py", line 273, in run
    sim.precond_solve(key_inlet = 'inlet.1',
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 861, in precond_solve
    self.solve(**kwargs)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1479, in solve
    self.OBJECTIVE_CYCLE(x0, self.x_state,
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1137, in OBJECTIVE_CYCLE
    self.one_cycle(X,
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1035, in one_cycle
    aborted = integrator.do_integration(N, tmin, tmax)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\integrators.py", line 139, in do_integration
    self.f1 = self.derivs(self.t0, self.xold)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 109, in derivs
    return self.sim.derivs(t, x)
  File "C:\ProgramData\anaconda3\envs\pdsimgb\lib\site-packages\PDSim\core\core.py", line 1632, in derivs
    self.core.properties_and_volumes(self.CVs.exists_CV, theta, STATE_VARS_TM, x)
  File "PDSim\core\containers.pyx", line 295, in PDSim.core.containers.CVArrays.properties_and_volumes
  File "PDSim\core\containers.pyx", line 342, in PDSim.core.containers.CVArrays.properties_and_volumes
  File "CoolProp\CoolProp.pyx", line 781, in CoolProp.CoolProp.State.update_Trho
  File "CoolProp\AbstractState.pyx", line 104, in CoolProp.CoolProp.AbstractState.update
ValueError: p is not a valid number
Thread-2: Process is done

[image]

Wang-Zongwei, 2023-08-27:
Attached is script file generated by PDSim GUI
[script_bdf61f53bc46.txt](https://github.com/ibell/pdsim/files/12448369/script_bdf61f53bc46.txt)
[script_ae3e98385bb1.txt](https://github.com/ibell/pdsim/files/12448370/script_ae3e98385bb1.txt)

ibell, 2023-08-28:
The short answer is don't use anything other than adaptive RK45. The others are not reliable, and to get them to be reliable enough you need an enormous number of steps, so it ends up still being faster to just use adaptive RK45.

Wang-Zongwei, 2023-08-29:
Clear! Thanks a lot for your answer.

closed 2023-08-29 by Wang-Zongwei