ibell/pdsim #85: PDSim GUI failed to use the vapor injection function

Wang-Zongwei, 2023-10-13:
Today I found below issue, and I hope my feedback is useful.

When I tried to use "vapor injection for scroll compressor", there's no any response when when I clicked the button "Add Injection Line". (see below pic)

[image]

Below information showed in main output log:
Traceback (most recent call last):
  File "C:\pdsim-master\GUI\plugins\scroll_plugins.py", line 422, in OnAddInjection
    IE = InjectionElementPanel(self.scrolled_panel,self.Nterms+1)
  File "C:\pdsim-master\GUI\plugins\scroll_plugins.py", line 265, in __init__
    IPP = InjectionPortPanel(self, index = 1)
  File "C:\pdsim-master\GUI\plugins\scroll_plugins.py", line 40, in __init__
    self.AddPort.Bind(wx.EVT_BUTTON,self.Parent.OnAddPort(self))
  File "C:\pdsim-master\GUI\plugins\scroll_plugins.py", line 340, in OnAddPort
    IPP = InjectionPortPanel(self, index = len(self.ports_list)+1)
AttributeError: 'InjectionElementPanel' object has no attribute 'ports_list'

ibell, 2023-10-18:
Hmm I haven't worked with that code in years. Can you debug yourself?

Wang-Zongwei, 2023-10-19:
OK, I will try, thanks.