ibell/pdsim #92: Coolprop 6.7.0 issue

Wang-Zongwei, 2025-03-12:
I hope my feedback is useful.

When using Coolprop 6.7.0, I found below issue and failed to start pdsim GUI: (when go back with Coolprop 6.6.0, no any issue for pdsim GUI)

Traceback (most recent call last):
  File "PDSimGUI.py", line 1953, in <module>
    frame = MainFrame()
  File "PDSimGUI.py", line 951, in __init__
    self.build()
  File "PDSimGUI.py", line 1195, in build
    self.MTB = MainToolBook(self, self.config)
  File "PDSimGUI.py", line 892, in __init__
    self.InputsTB = family.InputsToolBook(self, configdict)
  File "C:\python\externals\pdsim\GUI\families\scroll.py", line 186, in __init__
    pdsim_panels.StateInputsPanel(self, config['StatePanel'], name='StatePanel'),
  File "C:\python\externals\pdsim\GUI\panels\pdsim_panels.py", line 2659, in __init__
    state = CP.State(inletState.Fluid, dict(P=pressure,Q=1))
  File "CoolProp\\CoolProp.pyx", line 707, in CoolProp.CoolProp.State.__init__
  File "CoolProp\\CoolProp.pyx", line 740, in CoolProp.CoolProp.State.set_Fluid
TypeError: a bytes-like object is required, not 'str'

ibell, 2025-03-12:
I assume you use Python 3.13 but you didn't say. Please try the install from testpypi: ``pip install -i https://test.pypi.org/simple/ coolprop``

Wang-Zongwei, 2025-03-13:
I use Python 3.10 in Conda for pdsim when i found this issue.

Today I tried with the install from testpypi: pip install -i https://test.pypi.org/simple/ coolprop, but still with below failure:
I also tried to re-install pdsim in Conda with Python 3.13.2 + coolprop 6.7.1.post18, also same failure mode:

_Traceback (most recent call last):
  File "C:\pdsim\GUI\pdsimgui.py", line 1953, in <module>
    frame = MainFrame()
  File "C:\pdsim\GUI\pdsimgui.py", line 951, in __init__
    self.build()
    ~~~~~~~~~~^^
  File "C:\pdsim\GUI\pdsimgui.py", line 1195, in build
    self.MTB = MainToolBook(self, self.config)
               ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "C:\pdsim\GUI\pdsimgui.py", line 892, in __init__
    self.InputsTB = family.InputsToolBook(self, configdict)
                    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
  File "C:\pdsim\GUI\families\scroll.py", line 186, in __init__
    pdsim_panels.StateInputsPanel(self, config['StatePanel'], name='StatePanel'),
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pdsim\GUI\panels\pdsim_panels.py", line 2659, in __init__
    state = CP.State(inletState.Fluid, dict(P=pressure,Q=1))
  File "CoolProp\\CoolProp.pyx", line 707, in CoolProp.CoolProp.State.__init__
  File "CoolProp\\CoolProp.pyx", line 740, in CoolProp.CoolProp.State.set_Fluid
TypeError: a bytes-like object is required, not 'str'_

ibell, 2025-03-13:
> 1953, in frame = MainFrame() File "C:\pdsim\GUI\pdsimgui.py", line 951, in

Maybe I need to fix that one too. Can you please open an issue at CoolProp
repo and I'll fix?

Wang-Zongwei, 2025-03-13:
Thanks for your help. I will open an issue at CoolProp repo.

Wang-Zongwei, 2025-03-13:
[This issue has been opened at CoolProp repo.](https://github.com/CoolProp/CoolProp/issues/2497)

Wang-Zongwei, 2025-03-16:
I found this issue already solved with coolprop-6.7.1.post20. Thanks.

closed 2025-03-16 by Wang-Zongwei