ibell/pdsim #93: ValueError: critical point finding routine found 2 critical points

Wang-Zongwei, 2025-03-16:
[R454B_cofig.txt](https://github.com/user-attachments/files/19272181/R454B_cofig.txt)

I would like to report below issue for PDSim_GUI:

When I use fluid of R454B.MIX the first time in GUI, it's OK, no any issue.
But when the config file with R454B.MIX saved locally and then re-import it into the GUI, there're 2 issue:
 - The fluid cannot be changed again (Clicking the fluid area doesn't get any response from the GUI)
 - At the main output log: below error 

```
Traceback (most recent call last):
  File "C:\python\externals\pdsim\GUI\panels\pdsim_panels.py", line 2560, in UseChooser
    SCfrm=StateChooser(Fluid=Fluid,T=T,rho=rho,Fluid_fixed = self._Fluid_fixed)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\python\externals\pdsim\GUI\panels\pdsim_panels.py", line 2321, in __init__
    Tc = state.Props(CoolProp.iT_critical)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "CoolProp\\CoolProp.pyx", line 827, in CoolProp.CoolProp.State.Props
  File "CoolProp\\CoolProp.pyx", line 830, in CoolProp.CoolProp.State.Props
  File "CoolProp\\AbstractState.pyx", line 226, in CoolProp.CoolProp.AbstractState.keyed_output
ValueError: critical point finding routine found 2 critical points
```
I tried with below change as temporary solution:  (pdsim\GUI\panels\pdsim_panels.py)
```
line 2321, Tc = state.Props(CoolProp.iT_critical) --> Tc = state.Props(CoolProp.iT_max)
line 2323, pc = state.Props(CoolProp.iP_critical) --> pc = state.Props(CoolProp.iP_max)
```
But this is for sure not good solution.

Please help check if could fix it.  Thanks in [advance.]

[R454B_cofig.txt](https://github.com/user-attachments/files/19272336/R454B_cofig.txt)

[image]

[image]

Wang-Zongwei, 2025-03-16:
PDSim GUI version: 2.14
Coolprop version: 6.6.0

Wang-Zongwei, 2025-03-16:
Other fluid like n-Propane, R32, R410A, no this issue.
If call REFPROP fluid like REFPROP::R454B.mix, not having this issue either.
It seems all RXXX.mix will have this kind of issue.

ibell, 2025-03-16:
It depends on the mixture model. Multiple critical points are always possible for mixtures: https://scholar.google.com/citations?view_op=view_citation&hl=en&user=WNn0e_4AAAAJ&cstart=100&pagesize=100&sortby=pubdate&citation_for_view=WNn0e_4AAAAJ:IWHjjKOFINEC (that is the method implemented in CoolProp)

Wang-Zongwei, 2025-03-16:
Clear to me, thanks for your clarification.

closed 2025-03-16 by Wang-Zongwei