ibell/pdsim #99: State.State.copy() raising errors under CoolProp 6.7.0

xinding1207, 2025-04-02:
When updating the CoolProp to 6.7.0, the State.State.copy() function used in PDSim raises error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "CoolProp\\CoolProp.pyx", line 1136, in CoolProp.CoolProp.State.copy
  File "CoolProp\\CoolProp.pyx", line 1140, in CoolProp.CoolProp.State.copy
  File "CoolProp\\CoolProp.pyx", line 707, in CoolProp.CoolProp.State.__init__
  File "CoolProp\\CoolProp.pyx", line 740, in CoolProp.CoolProp.State.set_Fluid
TypeError: a bytes-like object is required, not 'str'
```

Seems like the CoolProp.State.copy function does not handle the arguements correctly.
Could you please help to provide guidance to solve this issue?

ibell, 2025-04-02:
Joy - looks like a bug in CoolProp. Can you please report there?

ibell, 2025-04-02:
And also report what python version and operating system you use?

xinding1207, 2025-04-02:
For sure! I will open an issue reporting this.